far-far-away-science/hab #5: Need to send APRS messages more frequently on decend closer to the ground

usaguerrilla, 2015-06-01:
subj

stcarlso, 2015-08-22:
Is this important? We want to avoid flooding the APRS system and every 30 seconds is already fairly often. Dithering was added in 75a7707 which should help.

usaguerrilla, 2015-08-28:
Might need to chat to Lee about this. I am not sure what's the protocol here.

stcarlso, 2015-10-08:
This is probably not necessary. Especially with the high signal quality of the PCB and the directional antenna build, increasing the signal frequency will probably just drain the battery faster on the ground.

Scorillo47, 2015-10-08:
I think we need this. The last seconds are the most important in transmitting the exact location. 30 second interval is too large.

usaguerrilla, 2015-10-10:
after talking to everybody we need to send messages with 10-15 seconds delay during last 1km of descend.
please note that base elevation in eastern Washington is quite higher than sea level.

commit 5811e309 (2015-10-11): Fixed #5: Need to send APRS messages more frequently on descend closer to the ground; Fixed #22: cleanup main.c
closed 2015-10-11 by stcarlso