far-far-away-science/hab #37: Log extra info to EEPROM (and maybe transmit it over the radio)

usaguerrilla, 2015-10-28:
- number of times we get watchdog timer reset
- number of different kinds of fault ISRs were triggered

stcarlso, 2015-10-30:
Writing to the EEPROM is a synchronous procedure which can take a large number of instructions, contrasting with devices such as the AVR which only take a few register writes. Performing this operation in a fault ISR or watchdog timer routine, where the system is probably already on the verge of crashing, will be infeasible.

It may be possible to examine the reset cause register, but this will not give fine grained information over the particular type of fault ISR thus triggered.

Scorillo47, 2015-10-30:
In V2 of the board, what if we add a SPI-based Flash chip next to the ARM? Something like W25Q64FVSSIG? We can log the information there 

http://www.digikey.com/product-detail/en/W25Q64FVSSIG/W25Q64FVSSIG-ND/2815931

stcarlso, 2015-11-02:
This would be better, but still tricky as we need to get the information out to the chip. We would have to pump the data out synchronously since interrupts would be disabled.

If we used a different ARM chip with true EEPROM in V2, such as an STM32 or Kinetis, this would be a non-issue as the data can simply be recorded in a few clock cycles.

Scorillo47, 2015-11-02:
Good point. Can we use a better ARM chip in that case?

usaguerrilla, 2015-11-12:
I totally forgot. We have a message counter. If we log all messages we can tell approximately how many times Tiva-C had been reset.

stcarlso, 2015-11-20:
Closing this issue for now. This functionality will be incorporated (if feasible) into the software for future flights.

closed 2015-11-20 by stcarlso