far-far-away-science/hab #42: migrate build to use gcc / eclipse instead of keil

usaguerrilla, 2015-12-02:
I added wiki just need to move code to use it

usaguerrilla, 2015-12-02:
might be a good idea to leave keil compiling (use trig functions there) as I can debug reliably only from it for now.
need to investigate further.

stcarlso, 2015-12-02:
Is this for Tiva or STM32 builds?

usaguerrilla, 2015-12-03:
Tiva

https://github.com/usaguerrilla/hab/issues/42#issuecomment-161457655

stcarlso, 2015-12-06:
This is probably not worth it for the Tiva builds unless you plan on updating the V1 Telemetry board after the first flight, or if you plan on maintaining a way to build the older code against an open source toolchain. Will leave the issue open until fixed for STM32 builds.

Scorillo47, 2015-12-06:
It's fine if we use Keil for the TIVA. I was using Code Composer, BTW, so far

GCC/Eclipse is a good common platform when we move to STM32

usaguerrilla, 2015-12-11:
You might be right. Might be a bad idea to change stuff which we know works fine.

usaguerrilla, 2016-03-28:
will do in v2

closed 2016-03-28 by usaguerrilla