far-far-away-science/hab #47: Assemble and test 2nd Tiva-C + Radio/Peripherals board

usaguerrilla, 2016-02-23:


stcarlso, 2016-03-04:
Assembled; testing still needs to happen. May be able to close it this weekend.

stcarlso, 2016-03-07:
I have found out that I was given a Stellaris LaunchPad instead of a Tiva-C version. Therefore, the code will not run. Do we have another Tiva?

Scorillo47, 2016-03-07:
I thought TIVA is just a brand rename of Stellaris - should have the same underlying SoC. 

I think Pavel has a TIVA

Scorillo47, 2016-03-07:
http://43oh.com/2013/04/ti-stellaris-launchpad-rebranded-to-tiva-c-launchpad/

Scorillo47, 2016-03-07:
BTW - if there is a hardware difference between the two, what would it take to port the code to Stellaris? 

I have a bunch of Stellaris boards but I have no TIVA-C

stcarlso, 2016-03-08:
On that page, it states that the Motion Control PWMs are new to the Tiva from the LM4F. The document at http://www.mouser.com/ds/2/405/lm4f120h5qr-124014.pdf lacks section 20, "Pulse Width Modulator (PWM)", so this module is unavailable on the LM4F Stellaris LaunchPad. This is a critical feature required for the APRS module, so we must use the Tiva C version.

Scorillo47, 2016-03-08:
Got it. I will order one TIVA

________________________________
De la: Stephen Carlson<mailto:notifications@github.com>
Trimis: ‎07.‎03.‎2016 18:14
Către: far-far-away-science/hab<mailto:hab@noreply.github.com>
Cc: Scorillo47<mailto:adi_oltean@hotmail.com>
Subiect: Re: [hab] Assemble and test 2nd Tiva-C + Radio/Peripherals board (#47)

On that page, it states that the Motion Control PWMs are new to the Tiva from the LM4F. The document at http://www.mouser.com/ds/2/405/lm4f120h5qr-124014.pdf lacks section 20, "Pulse Width Modulator (PWM)", so this module is unavailable on the LM4F Stellaris LaunchPad. This is a critical feature required for the APRS module, so we must use the Tiva C version.

---
https://github.com/far-far-away-science/hab/issues/47#issuecomment-193558353

stcarlso, 2016-03-12:
This has been fixed.

closed 2016-03-12 by stcarlso