far-far-away-science/hab #49: Helium cylinders case (so they don’t move when transported in the car)

usaguerrilla, 2016-02-23:


stcarlso, 2016-02-24:
How big does it need to be? Can this be made out of acrylic or wood on the laser?

usaguerrilla, 2016-02-24:
Adi has all the specs he measured them. I think it needs to be made from sturdy wood beams as cylinders are heavy. If you want you can chat to Adi and take this task from him.

Scorillo47, 2016-02-24:
I can make this easily at home - have all the materials.

usaguerrilla, 2016-02-24:
make sure you can do all items as you have many. if Stephen can help it would be awesome.

usaguerrilla, 2016-02-24:
Actually it can be done using laser cutter. Just need quite a bit of plywood. Basically mesh of plywood to support cylinders.

Scorillo47, 2016-03-03:
No. It can't be done with laser cutter. These are large 2x4s

Scorillo47, 2016-03-03:
Important - this task is very specific to my car and the props can't be reused except for a van of the same make/model (Honda Odyssey 2014) 

For a different car please open a separate task

usaguerrilla, 2016-03-03:
Depends on material. Plywood can be laser cut and assembled into a case. Need to research universal transporting solutions. Welders dont custom build this for each car.

Scorillo47, 2016-03-03:
It's totally OK from plywood it's just that it is more work. 
Adi

Date: Thu, 3 Mar 2016 13:58:22 -0800

Scorillo47, 2016-03-23:
Done!

closed 2016-03-23 by Scorillo47