far-far-away-science/hab #52: Build 5cm or 8cm balloon filling connector

usaguerrilla, 2016-02-23:


usaguerrilla, 2016-02-25:
Guys who sell balloons say HomeDepot has 5cm connectors.

Scorillo47, 2016-03-03:
So, Pavel, what is the exact diameter then of your balloon? 5 cm or 8 cm? Can you please check and then assign to me?

Scorillo47, 2016-03-03:
I have built the 3 cm connector + hose + quick connect so we can swap hose and/or cylinder. 

BTW, the connector should be either 3 cm (for Kayton HAB-800 ... HAB 1500) or 5 cm (for HAB-2000 or HAB-3000). Not 8 cm. 

http://www.kaymontballoons.com/Near_Space_Photography.html 

From our vendor data, it appears our balloon seems 3 cm, not 5 cm. Please double-check to make sure. 

I will close this for now, if it's 5 cm and not 3 cm, please open a separate task

usaguerrilla, 2016-03-04:
Ballon is 5cm.

Scorillo47, 2016-03-04:
OK, will build one as well

closed 2016-03-19 by stcarlso

Scorillo47, 2016-03-24:
Not sure why this was closed, but I've just finished building the 5 cm cylinder.

usaguerrilla, 2016-03-25:
Usually items are closed when they are done or no longer needed.