far-far-away-science/hab #57: Test final assembly including radio (as we have new antenna) from Bellevue City Center to Issaquah Highlands

usaguerrilla, 2016-02-23:


Scorillo47, 2016-03-03:
Need to think more about this. In the past antenna testing took a lot of time and was unclear what the success criteria was.

usaguerrilla, 2016-03-08:
We need to test if it works at all 1st. Then we can think about what to do next.

I will add a meeting for Thursday.

Test plan:
- test with SDR directional antenna to make sure signal can be received
- test with SDR omnidirectional large antenna to make sure signal can be received
- test with Kenwood directional antenna to make sure signal can be received
- test with Kenwood omnidirectional large antenna to make sure signal can be received

Send from City Center
Receive at Issaquah highlands

usaguerrilla, 2016-03-19:
done

closed 2016-03-19 by usaguerrilla