far-far-away-science/hab #59: Connect handhelds to PC (SDR, APRS decoder)

usaguerrilla, 2016-02-23:


stcarlso, 2016-02-27:
Do you have a recommended part number for an SDR if I wanted to try it out?

usaguerrilla, 2016-02-27:
You can check any. 

Here are cheap/good ones. I have 1st one.

http://www.amazon.com/NooElec-NESDR-Mini-RTL2832-Antenna/dp/B00P2UOU72/ref=sr_1_2?s=electronics&ie=UTF8&qid=1456597240&sr=1-2&keywords=sdr

http://www.amazon.com/NooElec-NESDR-Mini-Compatible-Packages/dp/B009U7WZCA/ref=sr_1_3?s=electronics&ie=UTF8&qid=1456597240&sr=1-3&keywords=sdr

stcarlso, 2016-02-27:
Is there a website listing a guide for how to set one up?

usaguerrilla, 2016-02-28:
Its not that difficult. You need to play with parameters to make sure you can receive APRS messages (there is a lot of traffic in western washington).

You can start by downloading SdrSharp software 

Though you might need a decent antenna for this.

At least you can listen to local radio stations using sdr.

stcarlso, 2016-03-12:
I got it working. Need to retest it outside to see if I can grab someone else's transmission, but I can receive and decode a signal.

closed 2016-03-12 by stcarlso

Scorillo47, 2016-03-12:
Wow, cool! 

Adi

Date: Fri, 11 Mar 2016 22:35:53 -0800