far-far-away-science/hab #64: Investigate if GoPro recorded anything and SD got corrupted or it simply stopped recording

usaguerrilla, 2016-03-28:


usaguerrilla, 2016-04-02:
1) Adi didn't find any extra files by searching SD card.

2) I was able to record 22GB worth of data into this card on my GoPro 2. Though this was after I formatted the card. Before that 2 out of 3 recorded files weren't playable.

Most likely GoPro was shut down during launch. Though it is possible there was some SD card trouble.

closed 2016-04-02 by usaguerrilla

Scorillo47, 2016-04-02:
We had issues with the previous card as well. Did we restart the GoPro?

________________________________
De la: Pavel Krupets<mailto:notifications@github.com>
Trimis: ‎02.‎04.‎2016 11:55
Către: far-far-away-science/hab<mailto:hab@noreply.github.com>
Cc: Scorillo47<mailto:adi_oltean@hotmail.com>
Subiect: Re: [far-far-away-science/hab] Investigate if GoPro recorded anything and SD got corrupted or it simply stopped recording (#64)

1) Adi didn't find any extra files by searching SD card.

2) I was able to record 22GB worth of data into this card on my GoPro 2. Though this was after I formatted the card. Before that 2 out of 3 recorded files weren't playable.

Most likely GoPro was shut down during launch. Though it is possible there was some SD card trouble.

---
https://github.com/far-far-away-science/hab/issues/64#issuecomment-204779446

usaguerrilla, 2016-04-02:
GoPro was powered off when I was swapping SD cards.

usaguerrilla, 2016-04-04:
GoPro2 might have overheated. I didn't observe any sudden movements in last seconds of video.
I got this reproduced at home.